Hexxeh/rpi-update #252: Boot fails after update when new kernel is installed and SD card encrypted

fis-cz, 2017-08-21:
Just would like to let you know that if initramfs is in use and rpi-update performed with a new kernel the initramfs is not updated and it is not possible to boot as there are no appropriate kernel modules to be used by cryptsetup.

It would be great if you can call some initramfs standard hook or all scripts in /etc/kernel/postinst.d/ folder with the installed kernel version as a only the one parameter. In this case it will be possible to write srcipts for refreshing the image if the new kernel is installed.

See https://robpol86.com/raspberry_pi_luks.html for details.

dsx724, 2019-03-28:
It would be great if this tool used /etc/default/raspberrypi-kernel's INITRD settings and update initrd as appropriate. That way this tool doesn't break booting for users that don't manually update initramfs. Kernel updates should go along with initramfs updates.

popcornmix, 2019-06-10:
See also: https://github.com/Hexxeh/rpi-firmware/issues/200

eljoest, 2020-03-19:
The regular kernel is calling something like ```run-parts -v --report --exit-on-error --arg=4.19.97-v7l+ --arg=/boot/kernel7l.img /etc/kernel/postinst.d``` after installation for each kernel image.
Custom hooks could then take care of the initrd generation. I've posted one to eljoest/rpi-initramfs-tools@394fd86ae6bece441d06b907442520f4e06579d9 Not sure if it's still useful.

Ruffio, 2020-12-28:
@fis-cz  can this be closed?

fis-cz, 2020-12-28:
Absolutely.

Ruffio, 2021-03-28:
@popcornmix  please close this per originator confirmation.

closed 2021-03-29 by popcornmix