Hexxeh/rpi-update #253: download fail,how to soulve this problem?

makelove, 2017-11-09:
```bash
pi@pi-desktop:~$ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 4.4.38-v7+
#############################################################
This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0     36      0 --:--:--  0:00:04 --:--:--    43
  5 54.1M    5 2831k    0     0   4942      0  3:11:20  0:09:46  3:01:34  6320
curl: (18) transfer closed with 53836053 bytes remaining to read

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
```
download 5%,it's downloading kernel.img,then stop
```bash
pi@pi-desktop:~$ sudo ls -alh /root/.rpi-firmware
total 4,0M
drwxr-xr-x 2 root root 4,0K Nov  9 15:22 .
drwx------ 8 root root 4,0K Nov  9 15:20 ..
-rw-rw-r-- 1 root root  16K Nov  3 21:02 bcm2708-rpi-0-w.dtb
-rw-rw-r-- 1 root root  16K Nov  3 21:02 bcm2708-rpi-b.dtb
-rw-rw-r-- 1 root root  16K Nov  3 21:02 bcm2708-rpi-b-plus.dtb
-rw-rw-r-- 1 root root  15K Nov  3 21:02 bcm2708-rpi-cm.dtb
-rw-rw-r-- 1 root root  17K Nov  3 21:02 bcm2709-rpi-2-b.dtb
-rw-rw-r-- 1 root root  18K Nov  3 21:02 bcm2710-rpi-3-b.dtb
-rw-rw-r-- 1 root root  17K Nov  3 21:02 bcm2710-rpi-cm3.dtb
-rw-rw-r-- 1 root root  50K Nov  3 21:02 bootcode.bin
-rw-rw-r-- 1 root root  19K Nov  3 21:02 COPYING.linux
-rw-rw-r-- 1 root root 2,6K Nov  3 21:02 fixup_cd.dat
-rw-rw-r-- 1 root root 6,4K Nov  3 21:02 fixup.dat
-rw-rw-r-- 1 root root 9,5K Nov  3 21:02 fixup_db.dat
-rw-rw-r-- 1 root root 9,5K Nov  3 21:02 fixup_x.dat
-rw-rw-r-- 1 root root   41 Nov  3 21:02 git_hash
-rw------- 1 root root 2,4M Nov  9 15:30 kernel.img
-rw-rw-r-- 1 root root 1,5K Nov  3 21:02 LICENCE.broadcom
-rw-rw-r-- 1 root root 714K Nov  3 21:02 Module7.symvers
-rw-rw-r-- 1 root root 720K Nov  3 21:02 Module.symvers
-rw-rw-r-- 1 root root  357 Nov  3 21:02 NOTICE.md
-rw-rw-r-- 1 root root  585 Nov  3 21:02 README.md
```

pi 3b
Ubuntu Mate 16.04
-
how to soulve this problem?
Network ?
I am from Beijing China
Thank you !

popcornmix, 2017-11-09:
> curl: (18) transfer closed with 53836053 bytes remaining to read****

This means you have network problems and cannot download a file from github reliably.
Are you using wired or wireless?
What is output when running:
```
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
```

makelove, 2017-11-10:
it is Working ,it's Network problems
Thanks

closed 2017-11-10 by makelove

wommy, 2017-11-11:
im having this same problem

`curl: (18) transfer closed with 10319265 bytes remaining to read

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now`

kswt, 2017-11-25:
I had the same error. I tiried many times, and every time it interrupts. It starts working after replacing the power supply to 5V 1.5 A instead of computer USB.
Raspberry Pi 3B.

FSKMTM, 2017-11-28:
This issue may be due to too slow on-the-fly untar of the downloaded archive. I have had this issue on the slower RPis (old Bs underclocked to 400 MHz) util I changed the download_rev function to first download the file and then untar it. The newer, faster RPis do not seem to have this issue.

ovdiyuk, 2017-12-03:
*** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    153      0 --:--:--  0:00:01 --:--:--   154
 84 54.1M   84 45.8M    0     0   247k      0  0:03:44  0:03:09  0:00:35  732k
curl: (18) transfer closed with 8726700 bytes remaining to read

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now

popcornmix, 2017-12-04:
@ovdiyuk What is output when running:
```
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
```

MrEngman, 2017-12-04:
@popcornmix 
I've seen a large number of similar errors over the last few weeks or so. Previously I don't recall any problems running rpi-update and I generally run rpi-update whenever there is a new commit to update the wifi drivers I compile and test them. 

Some errors. rather than saying they closed with a number of bytes remaining to read, appeared to refer to issues with tar. Unfortunately I did not save the error output and cannot find logs of the errors. Will the errors be logged somewhere?

Running `curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz` appears to be fine. Tried on the 6 Pis I have connected - 3B, 2x2B, B+, Zero and Zero W - without problems.

Something I've just noticed. When running your `curl` command the download runs smoothly without delays, but when running `rpi-update` the download occasionally appears to hang for short periods and this is when the errors may occur.

popcornmix, 2017-12-04:
> I've seen a large number of similar errors over the last few weeks or so

One question is what has changed? rpi-update itself hasn't changed for six months so doesn't seem to be the issue.
Other possibilities are something on github's end, or your ISP. I'd expect to see failures from the curl command too for those - are you sure the curl command is always reliable?

Another possibility is something in the kernel. That should be possible to diagnose.
You can force an rpi-update (even if you have that version) with:
```sudo rm /boot/.firmware_revision; sudo rpi-update <githash>```

From what you are saying a recent git hash will fail often, and a git hash from a couple of months ago will be reliable. If that is true then we can say the issue in in the contents of the rpi-update (presumably the kernel).

The other possibility is something in raspbian. Did it start to occur following an apt upgrade? Does starting from a clean raspbian image have the issue? What about an older clean raspbian image?

ovdiyuk, 2017-12-05:
@popcornmix 
I solved this following way 
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
SKIP_DOWNLOAD=1 rpi-update

popcornmix, 2017-12-05:
The exact command that is run (as root) by rpi-update is:
```
curl -L https://github.com/Hexxeh/rpi-firmware/tarball/master | sudo tar xzf - -C //root/.rpi-firmware --strip-components=1
```
Does running that as root fail for you? If so how reliably?

ovdiyuk, 2017-12-05:
I'm logged as root. Also I did try doing this with sudo. The same issue

ovdiyuk, 2017-12-06:
Raspbery Pi B+ model, OS Debian Jessie
SD card - for boot only
HDD - for system

curl works fine. But I had one interrupt while downloading. 
rpi-update was interrupted every time, It happened on different percentage of downloading (between 60%...95%). 
Yes, config.txt or cmdline.txt is changed to enable ATA0 for UART and others.
Unfortunate currently I have no access to copy text here. If you need exactly information what is in cmdline.txt I'll do it in couple days. 

I did run the curl command and the tar as two separate. Like below
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz

curl: (18) transfer closed with 8726700 bytes remaining to read
gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now

popcornmix, 2017-12-06:
> I did run the curl command and the tar as two separate. Like below
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
curl: (18) transfer closed with 8726700 bytes remaining to read
gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now

Okay, so you can get this error message both when running ``curl`` piped into ``tar`` and when running ``curl`` and ``tar`` as two separate commands.

If curl is failing on its own it does feel like an unreliable internet issue. Is the Pi connected via ethernet or wifi? Can you run the curl command on any other machines on the same LAN?

FSKMTM, 2017-12-06:
I did some testing here (IP 84.20.246.xxx) as well and I cannot reproduce the error by doing the download only. The curl download always comes through at the expecetd rate. It always fails, though, on a RPi model B if untar on the fly is attempted. If the RPi is clocked at default 700 MHz the transfer always stops at around 15% and the curl error 18 is thrown soon after. On a RPi 3 this does not happen. Both RPis are running Raspbian Stretch.

Gagravarr, 2017-12-31:
I've got a RPi model B, and a fairly fast internet connection. Like others, running `rpi-update` was failing for me every time with `curl: (18) transfer closed with ....` errors

I ran the curl manually, saving to a file, and that worked without errors. It also completed quite quickly. I then un-tar'd the `master.tar.gz` file, which took quite a lot longer to run than the curl did. I wonder therefore if the issue is that, on some older Pis with fast connections, the curl can hang trying to push data to tar, and the connection then dies while waiting?

I'd be tempted to at least add an option / env variable to download to a temp file then unpack, which those of us affected could then easily use. Maybe even on by default for older Pi versions with spare disk space?

reibuehl, 2018-01-16:
I have currently 4 RPis that show this exact issue. Some of them are connected via WiFi, some via Ethernet. On all of them I see these download errors at varying percentages of completion. I tried both, connecting directly and going via my local Squid proxy. When doing the manual, two step process as described by @ovdiyuk above, I also get the download errors but when downloading the tar from a windows PC in the same subnet, it works fine.

noequus, 2018-01-21:
I had the same problem as the OP; it failed after various levels of download completion (50-70%). I was able to run 
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz 
successfully, and it ran very quickly (I get about 20Mb/sec download speed from my ISP) 

Since I don't have the expertise to uncouple the download from the un-taring...based on the comments I tried throttling down the download speed via a router option. First attempt I went very slow...0.3Mb/sec...the download still failed, though earlier with only about 10% completion.

Perhaps that was TOO slow, causing another problem so as a last ditch effort I tried 1Mb/sec and the download worked and firmware was updated.  

I don't know with 100% certainty that the throttling was the reason--it is possible that whatever the problem was resolved itself anyway. But I've been struggling with these consistent download failures for more than a day and about 20 attempts, and the most recent failure was just before the one that worked. 

Perhaps this is helpful for others to try if they are having this irritating download problem. Thanks to those above for sharing their thoughts and expertise--being aware of the piping and a possible disconnect between the download and un-tarring was key to trying throttling the download.

ToninoTarsi, 2018-01-23:
I have exactly the same problem on an old RaspberryPI

RichiG88, 2018-01-25:
I had the same problem after I upgraded my RPi 3B (connected via wifi and running a f2fs filesystem) to debian stretch. After I disabled predictable network interface names in raspi-config and a reboot, rpi-update worked without errors.

popcornmix, 2018-01-26:
Can anyone else having this problem try disabling predictable network names?
I suspect this may just be random rather than a real workaround but would be useful to confirm.

ToninoTarsi, 2018-01-26:
With my old Raspberry PI 2 (System revision (000f) : Raspberry Model B PCB 2.0) I have the problem 100%. I was never able to update.
 So I taken the same SD and put it into a Raspberry PI Zero W and worked fine.

Atrixium, 2018-02-11:
I ran into this same issue with my older Pi 1B and had to run through the same process as ovdiyuk in order to update. It's worth emphasizing (for people with limited experience with tar, like myself) that you need to use the --strip-components=1 parameter (as mentioned by popcornmix) when decompressing the tarball so that rpi-update can actually find the data where it expects to, there is a leading directory in the tar that is not needed here and this strips it out.

mitesh10, 2018-02-19:
I was having the same problem albeit on a wireless connection and I switched over to wired and was able to download the file.

plumpsack, 2018-02-19:
rpi-update was failing for me every time too with "curl error 18".
 
The reason is the disk space on the boot partition. 
Old versions:
 
/dev/sdb1        8192    92159    83968   41M  c W95 FAT32 (LBA)

New, after re-partition:
/dev/mmcblk0p1        2048   217087   215040  105M  c W95 FAT32 (LBA)

uname -a
xxx 4.14.18+ #1093 Fri Feb 9 15:07:36 GMT 2018 armv6l GNU/Linux

jgottlander, 2018-03-07:
I deleted the /root/.rpi-firmware folder and then rpi-update worked again.

kregory, 2018-03-07:
juste resolv it by creating file ~root/.curlrc containing the following line
--keepalive-time 2

robrackn, 2018-03-15:
I've been experiencing this issue for a while running my RasPi 2 on a wired ethernet connection in Ubuntu bionic.

How would I manually download a specific kernel commit using curl? Once I extract it into /root/.rpi-firmware, what do I do next? I've only done rpi-update <commit> in the past and kernel upgrade have been automatic.

popcornmix, 2018-03-16:
@robrackn have you tried the keepalive suggestion in the post preceding yours?

mewashin, 2018-03-17:
@kregory's keepalive suggestion worked for me.

perrycannon, 2018-03-23:
@popcornmix
I solved this following way
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
SKIP_DOWNLOAD=1 rpi-update

good evening, how do I change direcotry into /root with raspbian on a raspberry pi?

Thank you Perry

popcornmix, 2018-03-23:
@perrycannon Have you tried the suggestion from @kregory ?
```echo "--keepalive-time 2" | sudo tee -a ~root/.curlrc```
should do it.

perrycannon, 2018-03-23:
Good evening. Yes I did as @kregory suggested. Below is the result. The transfer closed. I am working on a raspberry pi zero v1.3. I have previously performed rip-update on this pi zero with no issues.
$ curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current                                 Dload  Upload   Total   Spent    Left  Speed100   127    0   127    0     0    122      0 --:--:--  0:00:01 --:--:--   122100 14.0M    0 14.0M    0     0  77300      0 --:--:--  0:03:09 --:--:-- 88447curl: (18) transfer closed with outstanding read data remaining

robrackn, 2018-03-31:
@popcornmix Yes, I tried that. I still have the issue. I can manually download file, I think. Does placing the uncompressed files in the .rip-firmware folder automatically update the kernel on reboot?

popcornmix, 2018-04-02:
> Does placing the uncompressed files in the .rip-firmware folder automatically update the kernel on reboot?

No

d4nyll, 2018-07-14:
This is most likely because you have a `~/.curlrc` file with the `-w` option set. If set, this will append strings to `stdout`, after displaying your input. If you then pipe that directly to `tar` or `gzip`, it will think the file is corrupt.

Specifying an output file (with `-o` or `-O`) works because the string output by `-w` goes only to `stdout`, and not to the file itself. In the case of the pipe, however, they are one and the same.

To fix it, simply override your `.curlrc` and use `curl -w ""`.

popcornmix, 2018-12-05:
Latest rpi-update should include @d4nyll's suggestion

zeraien, 2019-11-01:
The best would be if the updater would cache the partially downloaded file and try again, because sitting in a country like Timor-Leste with 100kb/s download speeds, the download fails all the time, and has to restart, which is a nightmare. Have to do it manually.

kikislater, 2020-02-21:
Doing this thing manually with 
```
wget -c https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
SKIP_DOWNLOAD=1 rpi-update
```

watershade, 2020-10-31:
> @popcornmix
> I solved this following way
> curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
> cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
> SKIP_DOWNLOAD=1 rpi-update

Yes, it worked.
I choose download this package from my PC using vpn.(It is the key part), then use scp (you can choose copy by flash-disk )
~~~~~~~
# my package named rpi-firmware-master.tar.gz at ~/Downloads
sudo -i
cd /root
tar -xvzf /home/pi/Downloads/rpi-firmware-master.tar.gz .
rm -rf .rpi-firmware
mv rpi-firmware-master/ .rpi-firmware/
# I don't know why, but self-update is always fail, so...
SKIP_DOWNLOAD=1 UPDATE_SELF=0 rpi-update
exit
~~~~~~~

liuyanghan, 2020-12-30:
> 手动执行此操作
> 
> ```
> wget -c https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz
> cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
> SKIP_DOWNLOAD=1 rpi-update
> ```

root@raspberrypi:~/.rpi-firmware/rpi-firmware-master# SKIP_DOWNLOAD=1 rpi-update
 *** Github API is currently rate limited - please try again after . 

I used the same method, but it went wrong

pelwell, 2020-12-30:
The GitHub rate limiting can usually by bypassed with a GitHub access/API token: https://docs.github.com/en/free-pro-team@latest/github/authenticating-to-github/creating-a-personal-access-token

The `GITHUB_API_TOKEN ` environment variable tells `rpi-update` to use an API token: https://github.com/Hexxeh/rpi-update#github_api_token

satmandu, 2023-12-22:
FYI this would I think, be the current version of this:
```
sudo -s
mkdir -p /root/.rpi-firmware
curl -L https://github.com/raspberrypi/rpi-firmware/archive/master.tar.gz -o /root/master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
SKIP_DOWNLOAD=1 rpi-update
```

But this does not get around the ` *** Github API is currently rate limited` error... (And yes I know that the easiest way to deal with this is to wait or authenticate to Github.)

pelwell, 2023-12-22:
You'll have to complain to GitHub about this and/or use an API token - there's nothing we can do.