Hexxeh/rpi-update #260: Allow retry firmware download on failure

SkyLined, 2018-03-06:
I have a pi that needs to connect to the internet through a proxy. For unknown reasons the firmware download fails halfway through quite often when I using this proxy. If rpi-update had a switch that would allow it to retry the download until it succeeds, I would not manually have to run the script again and again to do so.

akhepcat, 2018-03-06:
Most instances of curl could be replaced by wget, which has a cleaner implementation of retry + continue.

There's really only one instance in the script that needs to be curl, and that's the line that grabs just the headers.

lurch, 2019-06-08:
You could try adding the `--retry NUM` option to the `curl` commands, to see if that helps? (but if you make local edits to `rpi-update`, don't forget to run it with `sudo UPDATE_SELF=0 rpi-update` to stop it overwriting your edits!)

lurch, 2019-06-11:
Now that #279 has been merged, you could do this with just `sudo CURL_OPTIONS="--retry 10" rpi-update`
:smiley: 
(although I've not tested this, as I have reliable internet!)

Ruffio, 2021-03-19:
@SkyLined can this issue be closed?

SkyLined, 2021-03-19:
Yes on my account - I no longer use a proxy, so I no longer need this feature and cannot test to see if the proposed solution works.

closed 2021-03-19 by SkyLined