Hexxeh/rpi-update #261: Invalid git hash specified

SkyLined, 2018-06-11:
I just installed a clean Raspbian Jesse Lite, made a few minor changes and wanted to update the firmware when I got this error:
```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 4.14.34-v7+
#############################################################
This update bumps to rpi-4.14.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689
##############################################################
Invalid git hash specified
```
My other pi's are up to date, so I cannot reproduce this issue on other machines until a new version comes out.

The error appears to come from this line of code:
https://github.com/Hexxeh/rpi-update/blob/551fe49161a7a21c6074de63f6891d15777346ce/rpi-update#L284

It wasn't immediately obvious to me what the problem is.

SkyLined, 2018-06-12:
This problem appears to have been temporary as it no longer reproduces.

closed 2018-06-12 by SkyLined

DaveDave2013, 2019-05-05:
Also getting this on a clean install of stretch.......

ricktendo, 2019-05-07:
Just received this same error.

pucelev, 2019-05-11:
Tried once more this morning and got the same error

peatrick, 2019-05-11:
I'm experiencing identical issues, on two out of three of my Raspberry Pi's:

- Raspberry Pi **3 Model B Plus** Rev 1.3, running: Raspbian GNU/Linux 9.9 (stretch) armv71 (_Kernel: 4.14.98-v7+_)
- Raspberry Pi **Zero W** Rev 1.1, running: Raspbian GNU/Linux 9.9 (stretch) armv61 (_Kernel: 4.19.34+_)

The only one that successfully updated was the Raspberry Pi 3 **Model B** Rev 1.2, running: Raspbian GNU/Linux 9.9 (stretch) armv71 and is now on (_Kernel: 4.19.40-v7+_).  Not sure why this one managed to properly update, while the other two are stuck.  If there's any further information I have left out, which may be beneficial in attempting to resolve this, please just say the word and I'm happy to try and provide logs, or what not.  Thank you kindly.

popcornmix, 2019-05-13:
Are you running pi-hole or similar dns based adblocker?
See: https://github.com/Hexxeh/rpi-update/issues/277