Hexxeh/rpi-update #262: rpi-update

dissipator, 2018-06-12:
I update to 4.14.48
but no kernel head in my system when i install  seeed-voicecard -v 0.3
dkms add -m seeed-voicecard -v 0.3
DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.14.48+ cannot be found at
/lib/modules/4.14.48+/build or /lib/modules/4.14.48+/source.

kregory, 2018-06-12:
You Have to install kernel-header

sudo apt-get install kernel-header

dissipator, 2018-06-13:
yes only 4.9.+ header

popcornmix, 2018-06-13:
Are you using an updated raspbian stretch distribution? That should be on 4.14.
You may also want to consider this: https://github.com/notro/rpi-source/wiki

dissipator, 2018-06-14:
OK ,I will try.thanks.......

dissipator, 2018-06-15:
*** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
/bin/sh: 1: scripts/basic/fixdep: Permission denied
scripts/Makefile.host:102: recipe for target 'scripts/basic/fixdep' failed
make[2]: *** [scripts/basic/fixdep] Error 126
Makefile:463: recipe for target 'scripts_basic' failed
make[1]: *** [scripts_basic] Error 2
make: *** No rule to make target 'modules_prepare'。 停止。
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 379, in <module>
    sh("cd %s && make modules_prepare %s" % (linux_symlink, (" > /dev/null" if args.quiet else "")))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 186, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /app/root/linux && make modules_prepare ' returned non-zero exit status 2

use the root user.

dissipator, 2018-06-15:
the same as:
https://github.com/raspberrypi/linux/issues/2579

Ruffio, 2020-12-07:
@dissipator is this issue still relevant?

Ruffio, 2021-03-19:
This issue is stall. Please consider to close.

closed 2021-03-19 by popcornmix