Hexxeh/rpi-update #267: rpi-update fails with "$1 ist nicht gesetzt"

gicking, 2018-10-28:
recent FW update of my NAS via "`sudo rpi-update`" fails with error message

`/usr/bin/rpi-update: Zeile 364: $1 ist nicht gesetzt.`  (translates to "line 364: $1 undefined" or so)

Line 364 in rpi-update reads: 

`local REPO_API=${REPO_URI/github.com/api.github.com\/repos}/commits/$1`

Last time everything worked fine, and in the meantime I only performed some `upgrade` / `dist-upgrade` via apt. Reboot didn't help and internet search didn't give any useful result. 

The configuration is a Pi3 running Raspbian Stretch (4.14.71-v7+) and some NAS services. Nothing fancy, no low level tinkering.

Any idea what goes wrong? Thanks a lot in advance!

popcornmix, 2018-10-28:
can you report output of:
```ls -l /boot/.firmware_revision```

gicking, 2018-10-28:
`ls -l /boot/.firmware_revision` --> 
`-rwxr-xr-x 1 root root 0 Okt 27 10:56 /boot/.firmware_revision`

`cat -l /boot/.firmware_revision` -->
no output (file has 0 bytes, see above)

I guess that is not correct...