Hexxeh/rpi-update #270: rpi-update fails downloading firmware due to redirect

BMicraft, 2018-12-05:
Running `rpi-update` on my up to date raspberry pi zero fails at un-gzipping the downloaded firmware. 
```
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    188      0 --:--:-- --:--:-- --:--:--   188

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```
When investigationg I added `set -o xtrace` in '/usr/bin/rpi-update' and ran it with updates disabled.
The offending line seems to be `eval curl -w "" -L https://github.com/Hexxeh/rpi-firmware/tarball/eefe4b161f5e9730183c8dc1605e14c85b15cf51`.
After eval that line get executed as `curl -w -L https://github.com/Hexxeh/rpi-firmware/tarball` and the empty format string for `-w` gets left out. That makes `-L` the new format string and therefore gets ignored and curl does not follow the redirect
```
# curl -w -L https://github.com/Hexxeh/rpi-firmware/tarball/eefe4b161f5e9730183c8dc1605e14c85b15cf51
<html><body>You are being <a href="https://codeload.github.com/Hexxeh/rpi-firmware/legacy.tar.gz/eefe4b161f5e9730183c8dc1605e14c85b15cf51">redirected</a>.</body></html>-L
```

A solution to that problem could be to wrap the empty quotes with another pair of single quotes
```
eval curl -w '""' -L ${GITHUB_AUTH_PARAM} "${REPO_URI}/tarball/${FW_REV}" | tar xzf - -C "${FW_REPOLOCAL}" --strip-components=1
```
By doing that the update worked flawlessly

popcornmix, 2018-12-05:
I've added the suggestion. Does it seem good now?

BMicraft, 2018-12-05:
Wow, that was fast.
I am new to github but I think just sent a pull request doing that, which that won't be necessary now.

closed 2018-12-05 by BMicraft