Hexxeh/rpi-update #271: gzip: stdin: not in gzip format, tar: Child returned status 1, tar: Error is not recoverable: exiting now

jmar83, 2018-12-05:
is it a bug on the kerne repo (=corrupted file that will be downloaded to my raspi) or is it a bug on the "rpi-update"-executable (=client) ??

```
root@xxx:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
This update bumps to rpi-4.14.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    251      0 --:--:-- --:--:-- --:--:--   251

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```

more here: https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=228370&p=1400867

jmar83, 2018-12-05:
"I think the "bug" is a corrupted file on the kernel repository, i always used "rpi-update" for many kernel version - .78, .79., 80., 82... now it stopped working...

"not in gzip format" -> i think that means the downloaded file can't be expanded because it's corrupted. so the bug is not on "rpi-update" executable (=client), it's on the server..."

(https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=228370&p=1400867)

popcornmix, 2018-12-06:
Is this still an issue? I think the problem was caused by an update to rpi-update script which was reverted last night.

jmar83, 2018-12-06:
Now it works, with .86 kernel, thx.

closed 2018-12-07 by popcornmix