Hexxeh/rpi-update #272: User confirmation (and a warning)

Jasata, 2018-12-12:
(please see now-closed issue #158)
Given that `rpi-udate` is included in a Rasbian distribution (2018-11-13, and I am sure in others as well) **and** because one of the most used commands has some resemblance to this (`raspi-config`) - would it be possible to have a simple prompt and a warning to anyone who fires up this utility.

Currently, it simply goes about its business and updates the firmware. It will do that even if you try to invoke it with options; `rpi-update --help` (this feels especially wrong).

The target audience of Raspberry boards should not be expected to know these things with confidence, and with tab-completion, this _does_ happen. For teachers (and club instructors) everywhere, please consider adding a user confirmation with warning messages (and perhaps `-y` option to bypass those, in case this command is also run by other scripts).

MrEngman, 2018-12-12:
Running rpi-update --help doesn't load any firmware. It updates itself but then gives an error.
...
pi@Pi-3B-plus:~ $ sudo rpi-update --help
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Invalid hash given
...

Ruffio, 2020-12-28:
When using rpi-update you read these lines:

```
root@raspberrypi:~# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Backing up modules 4.19.57+
#############################################################
WARNING: 'rpi-update' updates to pre-releases of the linux 
kernel tree and Videocore firmware.

'rpi-update' should only be used if there is a specific 
reason to do so - for example, a request by a Raspberry Pi 
engineer.

DO NOT use 'rpi-update' as part of a regular update process.

##############################################################
Would you like to proceed? (y/N)
```

I don't think the maintainers are going to change that, but I do think that it covers you initial request.
@Jasata consider to close this 'issue', thanks :-)

Jasata, 2020-12-29:
> When using rpi-update you read these lines:
> 
> ```
> 
> I don't think the maintainers are going to change that, but I do think that it covers you initial request.
> @Jasata consider to close this 'issue', thanks :-)

You are right, that is hugely better than 2018 situation. And I am pleased that invoking help no longer rams the update in as a bonus. Yes, perhaps it is time to close this now.

closed 2020-12-29 by Jasata