Hexxeh/rpi-update #273: How to use the backup created by rpi-update?

mittal07abhilash, 2019-01-10:
When running rpi-update it shows that it is creating a backup. Where to find that backup file and how to use it to revert to previous firmware and kernel? I tried using this script on OSMC and ended up with a broken system. Please help!

popcornmix, 2019-01-11:
Assuming you were running rpi-update for the first time you should see something like:
```
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 5.0.0-rc1-v7+
```
You should find it's created a backup of ``/boot`` called ``/boot.bak`` and a backup of ``/lib/modules`` called ``/lib/modules.bak``
If you can ssh in you can copy the backup files back to original location and reboot and that may fix the issue. But you may have a better response asking on https://discourse.osmc.tv/ where I suspect this question has been asked before.

PanderMusubi, 2019-09-19:
See also https://github.com/Hexxeh/rpi-update/issues/287

Ruffio, 2020-12-28:
@mittal07abhilash can this issue be closed?

closed 2020-12-30 by mittal07abhilash