Hexxeh/rpi-update #274: [feature request] When rpi-update invoked from CLI, /var/run/reboot-required is not set?  Can it be?

dachshund-digital, 2019-02-15:
It would seem like a good idea to set /var/run/reboot-required if a reboot is required, no?  This would mirror the logic apt-get establishes.

popcornmix, 2019-02-18:
Does this file have any effect on a default install of raspbian?
i.e. does it get automatically displayed somewhere?

J0WI, 2019-04-24:
It's used by various monitoring tools

dachshund-digital, 2019-04-25:
I under stand why it is set when set, the question is... should it be set when rpi-update runs... IMHO it should be set.  Right now it is not set, so other than a text message on screen warning that the firmware requires a reboot, no notification is set.  This is rather bad form.  Monitoring tools can't tell if firmware is active or not because the standard notification method is not used, as you noted, it is considered a standard for reboot notification.

popcornmix, 2019-04-25:
> It's used by various monitoring tools

More specifics would be useful.

rpi-update is designed for raspbian users. What affect does this file have on the standard preinstalled packages? What packages that are optionally installed does it have an effect on?

Now on Ubuntu, if I ssh in I see the contents of ``/var/run/reboot-required`` echoed back.
But on raspbian that doesn't occur.

lurch, 2019-06-08:
Closely related to #263 ?

dachshund-digital, 2019-06-08:
Close?  If the /var/run/reboot-required file or /var/run/reboot-required.pkgs files are generated by rpi-update when a reboot if requried, yes.  The issue I pointed out was that rpi-update does not follow the standard method for reboot required notification, and it really should.  There are actually to files that are generated:

1) /var/run/reboot-required... if the file exists at the given path, a reboot should be done.  Many monitoring solutions leverage the existence of this file, as a notification or warning trigger.  Node-Red, various monitoring tools, etc. leverage this.

2) /var/run/reboot-required.pkgs... if a file that is created and lists the packages installed, that require a reboot.  So when rpi-update runs, and a reboot is required, not only is /var/run/reboot-required file created, but the /var/run/reboot-required.pkgs file is created, if it does not exist, and "rpi-update" should be in the file.  If the file does exist already, than "rpi-update" should be appended at the end of file.

Clearly the first file is a trigger notification file.  The second file provides detail on why the first exists.  Most Linux distributions, including debian variants rely on these files as noted above.

Ruffio, 2020-12-28:
It looks like this is a 'standard' feature for both Debian and Ubuntu:

https://linux-audit.com/check-required-reboot-on-debian-ubuntu-systems/#:~:text=If%20the%20file%20%2Fvar%2Frun,names%2C%20but%20the%20related%20packages.&text=In%20this%20example%20we%20see,used%20by%20the%20Linux%20kernel

https://www.cyberciti.biz/faq/how-to-find-out-if-my-ubuntudebian-linux-server-needs-a-reboot/

https://serverfault.com/questions/92932/how-does-ubuntu-keep-track-of-the-system-restart-required-flag-in-motd

It would be a nice feature to get implemented if an update has sucessfully taken place IMHO.