Hexxeh/rpi-update #275: Confirm all flag (for automation)

Daverkex, 2019-02-20:
Like apt-get install X -y this tool need a flag for automatic updates like Ansible o similar software.

I try to update Raspberry firmware and my Ansible play fail because rpi-update promp for a major update in the kernel.

popcornmix, 2019-02-20:
Try setting ``SKIP_WARNING=1`` in environment if you don't won't want to be prompted by warnings.
But be aware the warnings are there for a reason.

Daverkex, 2019-02-28:
Thanks @popcornmix 
SKIP_WARNGING=1 working for me.
Maybe is a good idea add this flag in the README?

I know that skip the warnings in general is a bad idea but this automation is part of a boostrap, clean installation and up to date. I need update if the warning is a new kernel. Any idea for only allow this condition with the rpi-update (out of arkane magic with the managamente tool)? I think that skip all warnings can be a problem.

lurch, 2019-06-08:
@Daverkex It's not clear what you mean by "I need update if the warning is a new kernel ... I think that skip all warnings can be a problem." ?

Daverkex, 2019-12-14:
Sorry for my late answer.
Skip all warning with only a flag ca be a problem. In my case I only need skip if the warning is a new kernel. The automation need fail if other unspected warnings happens.
For this reason I think that a good idea is add a special flag to skip the warning if is a new kenel and other for skip all.

lurch, 2019-12-16:
Ahhh. I don't think that the warning-mechanism has a "reason for the warning" flag that you seem to be looking for.