Hexxeh/rpi-update #277: Invalid hash given

GabrielKn8, 2019-05-03:
"Would you like to proceed?" does not accept "y".

mbaccia514, 2019-05-08:
Also had this issue on a (relatively) fresh install. After performing apt updates the rpi-update command aborts with "Invalid git hash specified". Gives the same error when specifying different hashes as well.

See also:
https://github.com/Hexxeh/rpi-update/issues/261

ricktendo, 2019-05-08:
Yes, this error still happening to me since yesterday.

pucelev, 2019-05-10:
I tried this morning and I'm also getting this error

Shadesh, 2019-05-12:
Please make sure you don't (DNS or whatever) block GitHub Domains. For me it was "codeload.github.com" which was blocked by PIHole.

That prevented a successful download of fresh firmware with rpi-update.

pucelev, 2019-05-12:
Thanks Shadesh!  I disabled Pihole and was able to do the rpi-update.

ricktendo, 2019-05-12:
Yes, it was Pi-Hole

popcornmix, 2019-05-13:
I've added  a warning in troubleshooting section of readme about pi-hole blocking.

I further todo would be to detect the dns block directly by the tool.

Can someone with pi-hole set up to block report the output of:
```curl https://api.github.com/repos/Hexxeh/rpi-firmware/commits/master```

ricktendo, 2019-05-13:
Yes it does allow api.github.com, only codeload.github.com was blocked as malicious by the following two block lists:

1) https://www.squidblacklist.org/downloads/dg-malicious.acl
2) https://hosts-file.net/emd.txt

FYI I get my Pi-Hole blocklists from the following site: https://firebog.net/

I use the ticked lists: https://v.firebog.net/hosts/lists.php?type=tick

peatrick, 2019-05-14:
I guess I'll also chalk this one up to Pi-hole related issues.  Disabled (both) Pi-hole(s) and adjusted **/etc/resolv.conf** to use `nameserver 1.1.1.1` instead of my Unbound server(s) at 127.0.0.1 / localhost and am now happily using:
> `Linux pihole2 4.19.40-v7+ #1217 SMP Tue May 7 17:22:08 BST 2019 armv7l GNU/Linux`

qyqsoft, 2019-05-19:
You have to whitelist **codeload.github.com** in your pi-hole. That solved the issue for me.

OliverGruenberg, 2019-07-12:
just wanted to update kernel to v4.19.57-v7+ via 
sudo rpi-update 6e787b87ace9f945f17a9022288e2758ed281cf7

results in
 *** Invalid hash given

no pi-hole running btw

OliverGruenberg, 2019-07-12:
got it: the provided git hash in
https://github.com/Hexxeh/rpi-firmware/blob/master/git_hash
was incorrect.
The correct one for kernel v4.19.57 is:
 f8060224e53e26cdaea5b9f6f010ac8c3692e2ea

popcornmix, 2019-07-12:
https://github.com/Hexxeh/rpi-firmware/blob/master/git_hash

is the git hash of the kernel used to build the image. e.g.
https://github.com/raspberrypi/linux/commit/6e787b87ace9f945f17a9022288e2758ed281cf7

It is correct.

billouetaudrey, 2020-05-15:
Hi same issue when I used cloudflare dns

Ruffio, 2020-09-08:
@GabrielKn8 has this issue been resolved?

GabrielKn8, 2020-09-08:
@Ruffio l gave up on this. So, no update on my end.

pelwell, 2020-09-08:
These aren't problems that rpi-update can solve.

closed 2020-09-08 by popcornmix

Zeguym, 2020-12-17:
> Also had this issue on a (relatively) fresh install. After performing apt updates the rpi-update command aborts with "Invalid git hash specified". Gives the same error when specifying different hashes as well.
> 
> See also:
> #261

I had the same problem and it was missing public_repo right in the github access token

maguilara, 2021-09-25:
What a terrible feeling when this happens.
 My "apt update && upgrade" failed for some reason and kernel modules didn't match the one that I had currently installed, and rpi-update-ing to the corresponding commit with the matching kernel version didn't work with this exact error. 
I had not DNS issues, my pihole went down with my docker (which is why I realized something was wrong with the upgrade) and I didn't use the pihole dns on my raspberry pi anyway. 
I managed to solve it by downloading the whole firmware repo to that exact commit, and replaced /boot/ directory with the /boot/ directory from the branch downloaded Rebooted and it took minute to load correctly, but it went fine. Hopefully this helps someone. 
I recommend taking a backup of your SD before doing this! I didn't do it, but it was a risk I took, doesn't need to be your case!

gatedfuzz, 2021-09-25:
> My "apt update && upgrade" failed for some reason and kernel modules didn't match the one that I had currently installed, and rpi-update-ing to the corresponding commit with the matching kernel [...]

Same thing just happened to me because my time was exactly 5 hours out of sync. Strange.

maguilara, 2021-09-25:
> Same thing just happened to me because my time was exactly 5 hours out of sync. Strange.

Well, mine wasn't out of sync AFAIK, last reboot was less than 2 days ago, not enough time for me to detect that it was out of sync. And I believe the system asks for new time periodically either way. But my timezone is UTC-5, not sure if that is related or not.

gatedfuzz, 2021-09-25:
> Well, mine wasn't out of sync AFAIK, last reboot was less than 2 days ago, not enough time for me to detect that it was out of sync. And I believe the system asks for new time periodically either [...]

its funny because i cant get mine to sync back now. i was about to reflash anyway after i spent the evening seeing how quickly i could break it, guess i didnt have to try that hard.