Hexxeh/rpi-update #281: Soften Pi4 check_partition for use in directory destination

macmpi, 2019-07-03:
ref https://github.com/Hexxeh/rpi-update/commit/882f5c188dce31365885c8ff23724749294bf8c1

Addition of check_partition is causing issues on usecases where rpi-update is used to build distributions: in such case destination may be a simple directory rather than a partition per say.
Could this check be done in a "softer" manner, to play nice with such usecases?

Would be nice if WANT_PI4 usage is explained in README.md expert options too.
Thanks for consideration.

volumio, 2019-07-05:
Made a pull request: https://github.com/Hexxeh/rpi-update/pull/282

Ruffio, 2020-12-25:
@popcornmix what is status on this?

Ruffio, 2020-12-28:
@macmpi isn't this scenario covered by the merge of #283 ?
Can this be closed?

Ruffio, 2021-03-28:
@macmpi can this issue be closed?

macmpi, 2021-03-28:
sure, thanks.

closed 2021-03-28 by macmpi