Hexxeh/rpi-update #285: self-update clobbers script if it fails

HankB, 2019-07-22:
Raspbian Buster
Pi 4 (4GB)
Running rpi-update causes it to replace itself with a server error. I tried to resolve this by reinstalling rpi-update and when I ran it, it produced the same result. See below.
```
pi@nova:~ $ sudo dpkg --purge rpi-update
(Reading database ... 154123 files and directories currently installed.)
Removing rpi-update (20140705) ...
pi@nova:~ $ sudo apt install rpi-update
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  rpi-update
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,408 B of archives.
After this operation, 45.1 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf rpi-update all 20140705 [4,408 B]
Fetched 4,408 B in 1s (5,023 B/s)         
Selecting previously unselected package rpi-update.
(Reading database ... 154119 files and directories currently installed.)
Preparing to unpack .../rpi-update_20140705_all.deb ...
Unpacking rpi-update (20140705) ...
Setting up rpi-update (20140705) ...
pi@nova:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    27  100    27    0     0      1      0  0:00:27  0:00:14  0:00:13     5
 *** Relaunching after update
/usr/bin/rpi-update: line 1: 500:: command not found
pi@nova:~ $ cat /usr/bin/rpi-update
500: Internal Server Error
pi@nova:~ $ 
```
This has happened before: https://github.com/Hexxeh/rpi-update/issues/140
The root problem seems not to be fixed (e.g. success of the download is not checked before overwriting the script.)

popcornmix, 2019-07-22:
Checking the error code with ``-f``
https://github.com/Hexxeh/rpi-update/blob/master/rpi-update#L81
is meant to handle github errors, but I'm guessing the 500 returned this outage didn't count.

Maybe I'll add a check that script starts with the expected:
``#!/bin/bash``
before continuing.

popcornmix, 2019-07-22:
I've pushed a potential fix for this issue. Hard to be certain when github is now working.
Please test that latest rpi-update is working for you.

HankB, 2019-07-22:
I ran this on a new install of Raspbian/Buster on a Pi 2 B and it seemed to work. I copied the script downloaded from Github to /usr/bin/rpi-update. It reported that it self-updated and seemed to install the same version (matching md5sum.) It otherwise seemed to work (as did Github.)
Thanks for the quick response.

closed 2019-07-22 by HankB