Hexxeh/rpi-update #288: reinstall firmware

actraiser30, 2019-09-23:
Hello

i want to reinstall the same firmware i have in my raspberry and not a newer/beta one

how can i do it? 

and how can i install a specific firmware revision?

thanks

popcornmix, 2019-09-24:
What version do you want?
If the current apt version, then
```
sudo apt-get install --reinstall raspberrypi-bootloader
sudo apt-get install --reinstall raspberrypi-kernel
```
If it's a specific historical rpi-update version, then look at:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to update to that version.

Ruffio, 2020-08-29:
@actraiser30 has the issue been resolved and this can be closed?

Ruffio, 2020-12-07:
this issue should be closed due to inactivity for three months

closed 2020-12-07 by popcornmix