Hexxeh/rpi-update #290: Updates are not displayed with JUST_CHECK=1 rpi-update

Nastras, 2019-12-01:
Hello,

I have the following problem.

when I run sudo JUST_CHECK=1 rpi-update I get nothing displayed except this:

` *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Firmware update required. No local hash to compare to.
`

If I do the same thing on another Pi, all updates will be displayed.

Can someone tell me why nothing is displayed?

Thanks!

popcornmix, 2019-12-02:
It compares the current rpi-update version to the latest version and prints out the commit messages between. If you have never run rpi-update then this doesn't make sense.

The local hash it is looking for is ``/boot/.firmware_revision``

Nastras, 2019-12-03:
Thanks for your answer.
This message occurred at first start and does not disappear.

Do you have any idea how I can fix the error?

Nastras, 2019-12-03:
I just looked up the directory /boot/.firmware_revision is not available in my system. I created it manually but the error still exists -.-

popcornmix, 2019-12-03:
How did you create it?

Nastras, 2019-12-03:
I created the folder with mkdir as root. Is that wrong?
[image]

popcornmix, 2019-12-03:
The file needs to contain the git hash of the last rpi-update.
This list of commits since the last run of rpi-update only makes sense on the second update.
If you haven't run rpi-update before we don't know what has changed since the last one.

Once you have run rpi-update once, you can use ``JUST_CHECK=1 rpi-update``

Nastras, 2019-12-03:
I started rpi update now but without installing it. Unfortunately it still doesn't work JUST_CHECK=1 rpi-update. 

I just want to see the upcoming firmware updates with rpi-update 😁?
[image]

Nastras, 2019-12-03:
Thank you for your help!

popcornmix, 2019-12-03:
Did you delete the directory you created ``/boot/.firmware_revision``?
That will stop rpi-update from being able to create the *file* ``/boot/.firmware_revision``

Nastras, 2019-12-03:
No, I haven't. But I did it now and did rpi-update again. The error still exists.

[image]
[image]

popcornmix, 2019-12-03:
Did you enter "y" when asked "Would you like to proceed (y/N)" ?
You should see a ``*** Downloading specific firmware revision (this will take a few minutes)`` message which I don't see in your screenshot.

Nastras, 2019-12-04:
Hello, I have now confirmed with y and installed the updates. Now it works.

Actually I didn't want to install any updates and I just wanted to see the updates.

Fortunately everything still works.

Thanks for your help and effort.

closed 2019-12-04 by Nastras

popcornmix, 2019-12-04:
To just see the updates without running rpi-update you can browse:
https://github.com/Hexxeh/rpi-firmware/commits/master

Nastras, 2019-12-04:
Thank you for the hint. I get the information by telegram push so I use rpi-update.