Hexxeh/rpi-update #296: specify board version (e.g. on RPI 3: rpi-update SD card for RPI 4)

wivaku, 2020-04-22:
I am trying to fix my SD card by loading the SD into a Virtual machine Raspberry Pi and running rpi-update for that mounted SD card.

```
sudo mkdir -p /mnt/rescue/{boot,root}
sudo mount /dev/sdb1 /mnt/rescue/boot
sudo mount /dev/sdb2 /mnt/rescue/root
sudo ROOT_PATH=/mnt/rescue/root BOOT_PATH=/mnt/rescue/boot SKIP_BACKUP=1 rpi-update
umount /mnt/rescue/{boot,root}
```

This works fine, but when I boot my Raspberry Pi 4 using that SD, I get a blinking green led: 4x long, 4x short.
This seems to [stand for](https://www.raspberrypi.org/forums/viewtopic.php?p=437084#p1485558): `unsupported board type`.

If that is indeed the case/cause, is there a way to specify which board type to use?

pelwell, 2020-04-22:
`rpi-update` doesn't pay any attention to the device it is running on - it performs the same actions on all Pis because RPi images are supposed to run on all Pis. This works because Pis load the parts of the firmware that they need, e.g. Pi 3B -> start.elf, Pi 4B -> start4.elf.

Your interpretation of the flash code is correct - the firmware is saying it isn't meant to run on the Pi 4, either because it's the wrong variant or because it's too old to support it. It might save some time if you say what kind of image you are trying to fix, and how it came to be corrupted.

wivaku, 2020-04-23:
Long answer to give you the details.
In short: when using rpi-update from virtual machine I get the blinking leds, when using rpi-update from RPI 4 I don't (however: it still does not boot, stuck during the boot process).

The SD card is created on RPI 4 (Raspbian Buster Lite). I started to reinstall default firmware:
`sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`.  
During this process power was interrupted, after which the RPI would not boot anymore (stuck on colored screen). I can still mount the partitions on the card.

I used two approaches:
1. virtual machine with Buster and mount the faulty SD card
2a. freshly installed SD card on the RPI 4 and mount the faulty SD card → use `rpi-update`
2b. freshly installed SD card on the RPI 4 and mount the faulty SD card → use `apt-get --reinstall`
3. overwrite the boot partition files with files from newly installed system 

1. virtual machine
On my Mac I created a virtual machine using [RPI Desktop](https://www.raspberrypi.org/downloads/raspberry-pi-desktop/). I mounted the faulty SD card and then ran the commands above.

After inserting the rescued SD I get the blinking green LED (4x long, 4x short).

2a. copy from freshly installed SD card - use `rpi-update`
When copying the files from a newly installed boot partition to the rescued SD, the system boots but is stuck after `usb 1-1.3.3: Manufacturer: Mitsumi Electronic` (and after 3 minutes: `random: crng init done`)

2b. same as above, but use `sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`
same as 2a

Result of RPI 4 SD card after running 1:
[image]

Result of newly installed RPI 4 SD card:
[image]

wivaku, 2020-04-23:
Long answer to give you the details.
In short: when using rpi-update from virtual machine I get the blinking leds, when using rpi-update from RPI 4 I don't (however: had to update cmdline.txt to specify correct PARTUUID).

The SD card is created on RPI 4 (Raspbian Buster Lite). I started to reinstall default firmware:
`sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`.  
During this process power was interrupted, after which the RPI would not boot anymore (stuck on colored screen). I can still mount the partitions on the card.

I used two approaches:
1. virtual machine with Buster and mount the faulty SD card
2. freshly installed SD card on the RPI 4 and mount the faulty SD card

1. virtual machine
On my Mac I created a virtual machine using [RPI Desktop](https://www.raspberrypi.org/downloads/raspberry-pi-desktop/). I mounted the faulty SD card and then ran the commands above. After inserting the rescued SD I get the blinking green LED (4x long, 4x short).

2. from freshly installed SD card
On the RPI 4, use freshly installed SD card and mounted faulty card, then use rpi-update.
The system boots but is stuck after `usb 1-1.3.3: Manufacturer: Mitsumi Electronic` (and after 3 minutes: `random: crng init done`)
Update: turns out cmdline.txt had the incorrect PARTUUID. Booted ok after using the correct one.

pelwell, 2020-04-23:
I don't have time to debug how the image was broken by your VM setup. Fortunately for you, if you get the 4+4 flash sequence then the problem is in the FAT32 partition, so it's easier to analyse.

If it were my problem I'd start by comparing the files on the updated good image with those on the insufficiently fixed image. Use `md5sum /boot/*` and `md5sum /mnt/rescue/boot`, and compare the output. If one or more of the .elf files doesn't match, use `strings start4.elf | grep VC_BUILD` to work out how old it is and perhaps how it came to be there.

wivaku, 2020-04-23:
Same here, no time to debug. 
Approach 2 worked for me (after updating the cmdline.txt).

closed 2020-04-23 by wivaku