Hexxeh/rpi-update #297: Current kernel 4.19.97. rpi-update wants to install >90 kernel starting at 4.14.37...

mdrmdrmdr, 2020-04-22:
I recently upgraded my Raspi from Stretch to Buster. The Kernel was upgraded to:

`Linux raspberry 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux`

When I run rpi-update, it displays >90 "Bump to" messages. Almost all of them below my current version starting with **4.14.37**. And only 7 between my current version and the most recent 4.19.115.

What is rpi-update doing here? Installing >90 new kernel one after another?

popcornmix, 2020-04-22:
rpi-update shows the list of commits from the last time you updated with rpi-update.
It doesn't know if you updated with apt in the interim.

The list of commits is information. You will only update to the single firmware/kernel at the head of the tree (which will contain all preceding commits).

mdrmdrmdr, 2020-04-22:
Ah, understood. Thanks for the information...

Ruffio, 2020-09-08:
@mdrmdrmdr please close this issue if it has been resolved. Thanks.

closed 2020-09-08 by mdrmdrmdr