Hexxeh/rpi-update #298: Issues with kernel 4.19.115-v7+ rpi-update on ubuntu-mate 16.04.2

mrusli70, 2020-04-22:
Having issues with rpi-update on kernel 4.19.115-v7+ on ubuntu-mate 16.04.2, after upgrade the new firmware update it cannot reboot and hdmi gives no signal. Now ubuntu-mate 16.04.2 cannot boot up at all. Having upgrading issues with reboot problems since kernel 4.19.115.

pelwell, 2020-04-22:
Which device(s) are you testing?

mrusli70, 2020-04-22:
Raspberry pi 3

popcornmix, 2020-04-22:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

mrusli70, 2020-04-22:
The problem is now my raspbery pi 3 cannot even boot up ubuntu-mate 16.04.2 on my SD Card.
It is total. Brick.

pelwell, 2020-04-22:
And that is why we advise people not to run rpi-update without good reason.

mrusli70, 2020-04-22:
anyway to recover the boot issues on ubuntu-mate 16.04.2???

pelwell, 2020-04-22:
Do you have a spare SD card and a USB card reader? If so, download a fresh image - the Lite build will do - and write it to the new card, then come back for instructions.

mrusli70, 2020-04-23:
pelwell i install another image of ubuntu-mate 16.04.2 on another SD card and have USB card reader. I try to copy the boot.bak to boot. Still it would not boot up. No Signal on HDMI. Please let me know the step to do. Thank you.

mrusli70, 2020-04-23:
pelwell, i have check, the /lib/modules.bak and lib/modules are the same. The only different is that the /lib/modules files have the latest rpi-update kernels. The PI_BOOT files are there. But the PI_ROOT boot files are not there. Even though I copy the boot.bak to boot. There is no HDMI Signal. There is no boot to the SD Card.

pelwell, 2020-04-23:
OK, we can try a reinstall with an older version, but note:

**There is no guarantee this will work, and if this fails then I'm out of suggestions - you'll just have to try and salvage the data as best you can.**

1. With the broken card in the card read plugged into the Pi you should end up with two mount points - something like `/media/pi/boot` and `/media/pi/root`. I'm going to use those paths in the these instructions, but you may need to change them to match reality.
2. On the https://github.com/Hexxeh/rpi-firmware/commits/master page you'll find a list of releases. Because you are on a Pi 3 I'm going to guess that there might have been a problem with the switch to the common firmware on Apr 16th, so the last commit before that is https://github.com/Hexxeh/rpi-firmware/commit/10f878d6798d22599e50c6e4eb0e619924ee0d68 on Apr 15th, which shortens to `10f878d`.
3. Run the following command:
```
$ sudo BOOT_PATH=/media/pi/boot ROOT_PATH=/media/pi/root rpi-update 10f878d
```
4. When that completes, shutdown the Pi, move the patched card from the card reader into the normal slot and apply the power. Finger crossing is optional.

mrusli70, 2020-04-23:
Hi pelwell,

It work like a charm! I managed to get it to work on single attempt. Your methods really works.

I have my other SD Card with the similar Ubuntu-Mate 16.04.2 distro. And run the command 
that you mentioned. 

$ sudo BOOT_PATH=/media/pi/boot ROOT_PATH=/media/pi/root rpi-update 10f878d

Make some modification to the path while access to my other SD Card. 

$ sudo BOOTH_PATH=/media/rusli/PI_BOOT ROOT_PATH=/media/rusli/PI_ROOT rpi-update 10f878d

And shutdown the Pi down and reboot the patch card reader in to the PI. And it is working! Woohoo! 

Thank you pelwell. 

I wonder why the latest kernel was not supported with the Rapsberry pi 3???? 

There were issues too with 4.19.115 and the 4.19.115-V7+ with the Raspberry pi 3. 

Once again Thank you. Very much appreciate your help pelwell.

pelwell, 2020-04-23:
rpi-update is a testing ground for new kernels and firmware. Since the launch of Pi 4 we have been building its firmwares from a separate version of the code, but this is a maintenance headache because most changes have to be made to both side. The master branch of the releases has just switched to building all firmware from the same source code to give it some more testing, and some people have reported problems - not as extreme as yours, usually - and they are being investigated.

It looks as though you were unlucky with the timing of your update, or perhaps the update just went wrong in some way, but it underlines the fact that rpi-update and its firmware releases should be used with caution, preferably after a backup, and ideally not at all.

popcornmix, 2020-04-23:
You may want to follow https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=267576 for more 
information about what is being tested.

We aren't currently aware of an issue that would explain what you have seen.

If you have a spare sdcard and some free time it may be useful to install raspbian, run rpi-update and see if you also see the same issue.
If not, then install ubuntu-mate 16.04.2 and run rpi-update see if you see the same issue.

mrusli70, 2020-04-23:
There are bugs with kernel 4.19.115 and 4.19.115-v7+ with the raspberry pi 3. As after doing the rpi-updates, I cannot do any reboot and the power is totally down. No Signal on my HDMI output. I done it with my new SD Card. I had the same problems. I got no choice but to register and sign up myself to the github and ask for help. That is my only hope. LoL! I just hope they fix those bugs before it release to the public. Once again thanks for your help. pelwell.

pelwell, 2020-04-23:
> I just hope they fix those bugs before it release to the public.

FYI, "they" is us (@pelwell, @popcornmix and others), so "they" are aware of your problem and won't make a full OS release until they think it is ready.

mrusli70, 2020-04-23:
Oh thanks for the infos popcornmix.
I sure am. Doing those testing. Thanks for the help guys. Very much appreciate it.

mrusli70, 2020-04-23:
LoL! Ah... programming errors eh? Hee hee hee....

mrusli70, 2020-04-23:
LoL! you got to save the word man... oh don't tell me you guys are not using the old pi 3...

pelwell, 2020-04-23:
Please close the issue if you are satisfied with the answers.

closed 2020-04-24 by mrusli70