Hexxeh/rpi-update #299: self-update breaks script when GitHub is down

mrankine, 2020-04-23:
Running `sudo rpi-update` results in a broken `rpi-update` script, when GitHub is responding with 500/503 errors.

For me at the moment, `https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update` responds `500: Internal Server Error`. This is a general GitHub problem, not specific to rpi-update.

The problem is that the script does not detect the 500 code, and overwrites itself with this error response. So it will remain broken, even when GitHub is up again. **Suggestion: it should only overwrite itself if the HTTP response is 200.**

(As a workaround to download a valid script again, use `apt install --reinstall rpi-update`.)

Log:
```
pi@aleph:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100    26  100    26    0     0      1      0  0:00:26  0:00:21  0:00:05     5
 *** Relaunching after update
/usr/bin/rpi-update: line 1: 500:: command not found

pi@aleph:~ $ cat /usr/bin/rpi-update
500: Internal Server Error
```

pelwell, 2020-04-23:
I'm surprised by that - the code at least appears to be checking that the result is a bash script. It's under investigation.

mrankine, 2020-04-23:
Thanks.

This was on a fresh Raspbian image (2020-02-13 buster lite) and I'd just run `apt update && apt dist-upgrade`.

popcornmix, 2020-04-23:
If the version in apt repo is older than 22 Jul 2019 it won't contain the sanity check
https://github.com/Hexxeh/rpi-update/commit/a0c5de1e32194ed14818c93e183ca6cbb076059a

@XECDesign how old is rpi-update in apt?

mrankine, 2020-04-23:
For me, apt reports the latest rpi-update version is `20140705`.

popcornmix, 2020-04-23:
That's very old. I believe the issue is resolve in code from here, but apt needs an update.

mrankine, 2020-04-23:
Great, thanks! Do I need to file an issue somewhere else to get apt updated?

popcornmix, 2020-04-23:
Let’s wait for a response from @XECDesign - he can get it updated.

XECDesign, 2020-04-24:
Sorry, it's getting harder to spot mentions in all the github emails. Updated rpi-update package should show up in the repo within an hour.

mrankine, 2020-04-24:
Great, I can see the repo is now on version `20200409` which contains a0c5de1e32194ed14818c93e183ca6cbb076059a.

Thanks! Closing.

closed 2020-04-24 by mrankine