Hexxeh/rpi-update #300: Question regarding "DO NOT use 'rpi-update' as part of a regular update process."

TwizzyDizzy, 2020-04-29:
I was wondering... why is this line included? As I understand it, rpi-update also updates the kernel so one would expect this to explicitly be part of the update as one does not want to use older kernels which may contain unfixed security issues.

The same applies to current RPI firmware.

Am I missing something?

Cheers
Thomas

pelwell, 2020-04-29:
These releases are frequent and largely untested. If you make a habit of installing the latest when it is released you run the risk of something not working, possibly resulting in data loss. It's better to wait for the relevant Raspbian package to be updated, or for the maintainers of your preferred distro to decide that it is good enough for regular users.

TwizzyDizzy, 2020-04-29:
Well... I get the risk part... but on the other hand there's also the risk of running an insecure kernel.

Or should I rely on the kernel maintainers to backport the security fixes to the others branches whereby the security fixes would also land in the kernel that is included in the `raspberrypi-kernel` package (when it comes to Raspbian)? That would also make sense, if that's the case.

Cheers
Thomas

pelwell, 2020-04-30:
Your chronology is a bit off. We're currently shipping Raspbian images based on 4.19, and the master branch of rpi-update releases is still on 4.19. That kernel was first released in October 2018. Meanwhile, 18 months later, the Linux devs are on 5.7-rc2. So the lag between top-of-tree kernel development and rpi-update release is _much_ larger than the lag between rpi-update and Raspbian.

If the only way to get security updates was to install the latest kernel then we'd all be in trouble. Fortunately, any commit with a `Fixes:` tag (and some that don't) gets back-ported to older kernels; the Long Term Stable kernels (of which 4.19 is one, and 5.4 is the next) get years of support. 4.19 is now on 4.19.118, while the most recent `raspberrypi-kernel` package is on 4.19.97 (from the end of January).

The discovery of a critical vulnerability will accelerate the usual kernel package update cycle, and the additional testing that goes with it. You really are better off sticking with `apt upgrade` and friends unless your intention is to help us spot issues early.