Hexxeh/rpi-update #301: Firmware 4.19.115-v7+ Bluetooth can't paired to any of my smart phone on Raspberry Pi 3B+

mrusli70, 2020-05-07:
Hi

I have a raspberry Pi 3b+. With the firmware update version 4.19.115-V7+.
I tried on my iphone and  samsung galaxy note 9. And it would not connect to bluetooth on my raspberry pi 3B+ with Raspbian OS. 

Even I used the bluetoothctl -a and pair xx:xx:xx:xx:xx:xx commands it does not work. 

It's either raspbian os is not working or the firmware is not working.

pelwell, 2020-05-07:
What are you expecting to be able to do with it once it is paired? The Raspbian Bluetooth support is written with the view of the Pi being the host driving external audio devices. Your phone is in the same position - wanting to be the source - so I think you're probably expecting to use the Pi to create a Bluetooth speaker. If you search for "use raspberry pi to make a bluetooth audio receiver" you will find a number of tutorials explaining how to do that.

mrusli70, 2020-05-07:
pelwell, i figure i have to install blueman and gnome-bluetooth to get it working on raspbian. 
It seems that it only work with the smartphone wifi for the android 10 and turn on the mobile data with personal hotspots on the iphone. There is no other way that I can get the it to surf the web on the mobile data from raspbian. Wifi bluetooth on iphone does not seems to work. 

What i really wanted to do is to use Internet Connection Sharing from Raspbian to my smart phone.

pelwell, 2020-05-07:
If you search for "raspberry pi tether to iphone over bluetooth" you'll find a number of tutorials, but it certainly isn't supported natively by Raspbian. You may also find that the performance isn't what you would like - the hotspot approach is likely to be much better.

mrusli70, 2020-05-08:
Thanks for the infos pelwell. 
Got some links that i stumble with your help. 
https://www.raspberrypi.org/forums/viewtopic.php?t=223029
https://github.com/WayneKeenan/RaspberryPi_BTPAN_AutoConnect
https://github.com/billz/raspap-webgui

closed 2020-05-09 by mrusli70