Hexxeh/rpi-update #303: Invalid hash given

jonealex, 2020-06-01:
Hello everyone,

Since a few weeks I can't update using `sudo rpi-update`, it throws the error `Invalid hash given`:

[image]

How can I solve it? Thank you in advance.

Kind regards.

pelwell, 2020-06-01:
Duplicate of #277.

jonealex, 2020-06-01:
**UPDATE:**

Changing DNS and using ISP defaults now I can "proceed" but results in the same error:

[image]

Note: I'm not using pi-hole.

popcornmix, 2020-06-01:
Can you report output of:
```
curl -s -L https://api.github.com/repos/Hexxeh/rpi-firmware/commits/master
```
and
```
wget https://github.com/Hexxeh/rpi-firmware/tarball/master
```

jonealex, 2020-06-01:
> Can you report output of:
> 
> ```
> curl -s -L https://api.github.com/repos/Hexxeh/rpi-firmware/commits/master
> ```
> 
> and
> 
> ```
> wget https://github.com/Hexxeh/rpi-firmware/tarball/master
> ```

**curl**:

```
pi@netbites:~ $ curl -s -L https://api.github.com/repos/Hexxeh/rpi-firmware/commits/master
{
  "sha": "b2aabc3ac7852481cb7833ce068f65eaa4476a2f",
  "node_id": "MDY6Q29tbWl0NDA5NTgxMTpiMmFhYmMzYWM3ODUyNDgxY2I3ODMzY2UwNjhmNjVlYWE0NDc2YTJm",
  "commit": {
    "author": {
      "name": "popcornmix",
      "email": "popcornmix@gmail.com",
      "date": "2020-06-01T12:32:13Z"
    },
    "committer": {
      "name": "popcornmix",
      "email": "popcornmix@gmail.com",
      "date": "2020-06-01T12:32:13Z"
    },
    "message": "firmware: il: isp: Ensure HR output is active and ISP is open before starting a frame\nfirmware: isp_ctrl: Fail in start_[raw|yuv]_frame if ISP is not idle\nSee: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=275489",
    "tree": {
      "sha": "f72b6e08bdaf4b2028e99567110014517f898623",
      "url": "https://api.github.com/repos/Hexxeh/rpi-firmware/git/trees/f72b6e08bdaf4b2028e99567110014517f898623"
    },
    "url": "https://api.github.com/repos/Hexxeh/rpi-firmware/git/commits/b2aabc3ac7852481cb7833ce068f65eaa4476a2f",
    "comment_count": 0,
    "verification": {
      "verified": false,
      "reason": "unsigned",
      "signature": null,
      "payload": null
    }
  },
  "url": "https://api.github.com/repos/Hexxeh/rpi-firmware/commits/b2aabc3ac7852481cb7833ce068f65eaa4476a2f",
  "html_url": "https://github.com/Hexxeh/rpi-firmware/commit/b2aabc3ac7852481cb7833ce068f65eaa4476a2f",
  "comments_url": "https://api.github.com/repos/Hexxeh/rpi-firmware/commits/b2aabc3ac7852481cb7833ce068f65eaa4476a2f/comments",
  "author": {
    "login": "popcornmix",
    "id": 1053730,
    "node_id": "MDQ6VXNlcjEwNTM3MzA=",
    "avatar_url": "https://avatars3.githubusercontent.com/u/1053730?v=4",
    "gravatar_id": "",
    "url": "https://api.github.com/users/popcornmix",
    "html_url": "https://github.com/popcornmix",
    "followers_url": "https://api.github.com/users/popcornmix/followers",
    "following_url": "https://api.github.com/users/popcornmix/following{/other_user}",
    "gists_url": "https://api.github.com/users/popcornmix/gists{/gist_id}",
    "starred_url": "https://api.github.com/users/popcornmix/starred{/owner}{/repo}",
    "subscriptions_url": "https://api.github.com/users/popcornmix/subscriptions",
    "organizations_url": "https://api.github.com/users/popcornmix/orgs",
    "repos_url": "https://api.github.com/users/popcornmix/repos",
    "events_url": "https://api.github.com/users/popcornmix/events{/privacy}",
    "received_events_url": "https://api.github.com/users/popcornmix/received_events",
    "type": "User",
    "site_admin": false
  },
  "committer": {
    "login": "popcornmix",
    "id": 1053730,
[... 171 lines cut ...]
      "contents_url": "https://api.github.com/repos/Hexxeh/rpi-firmware/contents/start_db.elf?ref=b2aabc3ac7852481cb7833ce068f65eaa4476a2f"
    },
    {
      "sha": "2bb7770d367911d611aaadbc43cb0c00982c9a3d",
      "filename": "start_x.elf",
      "status": "modified",
      "additions": 0,
      "deletions": 0,
      "changes": 0,
      "blob_url": "https://github.com/Hexxeh/rpi-firmware/blob/b2aabc3ac7852481cb7833ce068f65eaa4476a2f/start_x.elf",
      "raw_url": "https://github.com/Hexxeh/rpi-firmware/raw/b2aabc3ac7852481cb7833ce068f65eaa4476a2f/start_x.elf",
      "contents_url": "https://api.github.com/repos/Hexxeh/rpi-firmware/contents/start_x.elf?ref=b2aabc3ac7852481cb7833ce068f65eaa4476a2f"
    }
  ]
}
```

**wget**:

```
pi@netbites:~ $ wget https://github.com/Hexxeh/rpi-firmware/tarball/master
--2020-06-01 17:36:16--  https://github.com/Hexxeh/rpi-firmware/tarball/master
Resolviendo github.com (github.com)... falló: Nombre o servicio desconocido.
wget: no se pudo resolver la dirección del equipo “github.com”
```

popcornmix, 2020-06-01:
You've got a problem with your DNS lookups. It can't locate github.com (where we are now).
Either a bug with DNS provider, an invalid setting in your local lookup (e.g. /etc/hosts) or some form of adblocking is behaving badly. It's not something we can fix from here.

To narrow it down try the same wget command from another PC on your local network. Does it fail there too?

jonealex, 2020-06-01:
> You've got a problem with your DNS lookups. It can't locate github.com (where we are now).
> Either a bug with DNS provider, an invalid setting in your local lookup (e.g. /etc/hosts) or some form of [...]

Stranger things... on my main computer (Windows 10) I can resolve it without problems:

```
C:\Users\aleja>ping github.com

Haciendo ping a github.com [140.82.118.3] con 32 bytes de datos:
Respuesta desde 140.82.118.3: bytes=32 tiempo=49ms TTL=51
Respuesta desde 140.82.118.3: bytes=32 tiempo=49ms TTL=51
Respuesta desde 140.82.118.3: bytes=32 tiempo=54ms TTL=51
Respuesta desde 140.82.118.3: bytes=32 tiempo=49ms TTL=51

Estadísticas de ping para 140.82.118.3:
    Paquetes: enviados = 4, recibidos = 4, perdidos = 0
    (0% perdidos),
Tiempos aproximados de ida y vuelta en milisegundos:
    Mínimo = 49ms, Máximo = 54ms, Media = 50ms
```

But on Linux Subsystem (same computer, same local network):

```
jonealex@DESKTOP-DE9UPRV:~$ wget https://github.com/Hexxeh/rpi-firmware/tarball/master
--2020-06-01 17:44:45--  https://github.com/Hexxeh/rpi-firmware/tarball/master
Resolving github.com (github.com)... failed: Name or service not known.
wget: unable to resolve host address ‘github.com’
```

popcornmix, 2020-06-01:
On linux you can add the line:
```140.82.118.3 github.com```
to `/etc/hosts` which may work around the issue. But it's probably a problem with your router, or something outside of the Pi.

jonealex, 2020-06-02:
> On linux you can add the line:
> `140.82.118.3 github.com`
> to `/etc/hosts` which may work around the issue. But it's probably a problem with your router, or something outside of the Pi.

Yup, I solved by adding a simple line to `/etc/hosts` but I should talk about with my ISP.

Thank you !

closed 2020-06-02 by jonealex