Hexxeh/rpi-update #304: /boot/kernel7l.img needs to exist to get a V8 kernel installed

DougieLawson, 2020-06-03:
If `/boot/kernel7l.img` (which can't be used on RPi3B) doesn't exist then rpi-update doesn't even attempt to install the 64-bit kernel8.img.

This bit of code 
```
		find "${FW_REPOLOCAL}/modules" -mindepth 1 -maxdepth 1 -type d | while read DIR; do
			BASEDIR="$(basename "${DIR}")"
			if [[ ${WANT_PI4} -ne 1 ]]; then
				ENDSWITH=${BASEDIR: -4}
				if [[ "${ENDSWITH}" == "v7l+" ]]; then
					continue;
				fi
				ENDSWITH=${BASEDIR: -3}
				if [[ "${ENDSWITH}" == "v8+" ]]; then
					continue;
				fi
			fi
``` 
doesn't work on a RPi3B.

popcornmix, 2020-06-03:
I believe it is working as designed.

Older (pre-buster) installs had too small a FAT partition to add the the extra kernel and firmware files.
A decision was made that rpi-update wouldn't add new files to /boot without `WANT_PI4=1` being set.

`WANT_PI4=1` is implicitly set if you already have pi4 files on the /boot partition.

You can force it by setting `WANT_Pi4=1` once. From then on it will not be required.

I accept that kernel8.img isn't exclusively Pi4 (but it is mainly there for Pi4), so perhaps the control is imperfectly named, but I feel it is close enough.

DougieLawson, 2020-06-03:
I can't eliminate the kernel7l.img on 3Bs. If I do that the kernel8.img isn't installed. That's a bug not a feature. Try it.

If I could fathom the complexity of your shell script and had a 4B to test on I'd hack up a pull request.

satmandu, 2021-01-08:
Also seeing this issue. For what it is worth I needed to do a  ```touch /boot/kernel7l.img && sudo WANT_Pi4=1 rpi-update``` to update /boot/kernel8.img for me.

Ruffio, 2021-03-28:
Has this issue been 'resolved'?

DougieLawson, 2021-03-28:
I worked around this with `WANT_PI4=1` when running `sudo rpi-update`.

closed 2021-03-28 by DougieLawson