Hexxeh/rpi-update #306: PRUNE_MODULES=1 does not seem to do anything

kaoscoach, 2020-06-11:
I had a previous Pi running Raspbian that ran out of disk space and had to rebuild. Since my reinstall I have been very cognizant of checking its disk space. After a couple of rpi-update's I have tried running it with the PRUNE_MODULES=1 setting but I still see old versions in /lib/modules. What files is PRUNE_MODULES=1 supposed to be removing?

popcornmix, 2020-06-16:
See: https://github.com/Hexxeh/rpi-update/issues/197
For me it prints out:
```
Pruning /lib/modules/5.4.42+
Pruning /lib/modules/5.4.28-v7l+
Pruning /lib/modules/4.19.79-v8+
Pruning /lib/modules/4.19.79+
Pruning /lib/modules/4.19.23-v7+
Pruning /lib/modules/4.19.84-v7l+
Pruning /lib/modules/4.19.84-v7+
Pruning /lib/modules/4.19.57+
Pruning /lib/modules/5.5.0-rc5+
Pruning /lib/modules/5.5.0-rc5-v7l+
Pruning /lib/modules/4.19.60-v7+
Pruning /lib/modules/4.19.57-v7+
Pruning /lib/modules/4.19.1-v7+
Pruning /lib/modules/4.19.85-v8+
Pruning /lib/modules/4.19.80-v8+
Pruning /lib/modules/4.19.105-v7l+
Pruning /lib/modules/4.19.56+
Pruning /lib/modules/4.19.37-v7+
Pruning /lib/modules/4.19.64-v7l+
Pruning /lib/modules/4.19.102-v8+
```
It deletes any directory in `/lib/modules/*` whose content only include files of type `*.ko` or `modules.*` (i.e. default kernel modules)

kaoscoach, 2020-06-16:
I don't get that output or anything like that. I am using pi-hole on Raspbian. Is there something I should do to help troubleshoot?

popcornmix, 2020-06-16:
Can you show me what you entered to run rpi-update?

pelwell, 2020-06-16:
What exactly are you typing to run the prune?

kaoscoach, 2020-06-16:
sudo PRUNE_MODULES=1 rpi-update

kaoscoach, 2020-06-16:
Looking at #197 helped. I deleted the /root/.firmware_revision and re-ran, which worked. Is deleting /root/.firmware_revision a requirement?

popcornmix, 2020-06-16:
It deletes all modules before installing the new ones. It will only do this if you actually updating. Otherwise you'd be left with no modules at all.

kaoscoach, 2020-06-16:
So I performed the command incorrectly? As I said, if I run, sudo PRUNE_MODULES=1 rpi-update on an up-to-date system, nothing happens for me. Only worked after I deleted /root/.firmware_revision.

popcornmix, 2020-06-16:
It only deletes modules when it has an update to do. If you are up to date the script does nothing.

kaoscoach, 2020-06-16:
I guess I am confused. I thought running sudo PRUNE_MODULES=1 rpi-update on an up-to-date system would remove unneeded/unused files. But you are saying I have to run, sudo PRUNE_MODULES=1 rpi-update on an out of date system in order for it to work? Is there a way to do a uname -a and delete all that do not correspond to the current running kernel? At least that way it could remove unneeded/unused files on an up-to-date system while keeping those that are currently in use.

popcornmix, 2020-06-16:
The description of PRUNE_MODULES is "Allows you to delete unused module directories *when doing an update*."
You need to have an update to do for it to do anything. As you discovered you can force an update by removing ` /root/.firmware_revision`.
There are no plans to change the behaviour.
If you always use `PRUNE_MODULES=1` in the future when running rpi-update your sdcard space will no longer fill up with old modules.

kaoscoach, 2020-06-16:
But if I use PRUNE_MODULES=1 in the future, won't that delete a possible working version of my system? When I reboot and if I deleted the modules prior to the update, I would not be able to fix it since those files would be deleted? 

Initially, I wasn't sure which files were 'needed', so I guess if you do not plan to change the behavior, I will just script a way to delete the unneeded directories after comparing it to a current running 'uname -a'. Since I now know modules that do not match the uname are not needed. Thanks.

popcornmix, 2020-06-16:
`PRUNE_MODULES=1` only deletes modules before it installs new ones. When you reboot it will use the new modules that have been installed (along with the new kernel which only works with the new modules).

closed 2020-06-16 by popcornmix