Hexxeh/rpi-update #309: log file somewhere?

alanbork, 2020-09-01:
I've been testing new firmware and the current one breaks something I need - is there a way to tell which test firmware I had most recently so I could downgrade to that one (rather than going all the way back to the stock install)? I can't find any logs showing which branch rpi-update had installed, and I didn't save what it had printed out at the command line (it was a few weeks ago).

popcornmix, 2020-09-02:
rpi-update doesn't keep logs.

You may get some information from /var/log. e.g.
```
zgrep "Linux version" /var/log/kern.log*
```
I have a number of older log files, e.g. kern.log.1.gz-2019102900.backup which I can extract the kernel version from.

alanbork, 2020-09-02:
Unless I'm missing something that doesn't tell you the github commit ####
is, which is the part you actually need to roll back.

pelwell, 2020-09-02:
Yes, but the kernel version allows you to search for a commit which is at least close, and sometimes a precise match.

pelwell, 2020-09-02:
And when I say search I mean on this page: https://github.com/Hexxeh/rpi-firmware/commits/master

Ruffio, 2020-12-28:
@alanbork can this issue be closed?

alanbork, 2020-12-28:
It's still not fixed, would be easy to implement, and would be a significant improvement to program to log which version is being installed somewhere. So no, I don't think it should be closed.

alanbork, 2021-07-03:
you can find the currently installed branch in /boot/.firmware_revision

not sure if this is a feature addition or just unknown when this issue was opened,.

closed 2021-07-03 by alanbork