Hexxeh/rpi-update #310: ssl error when runnung rpi-update on rpi3

kamomil, 2020-11-08:
hi, I have rpi3 woth raspbain,
when running `rpi-update` I get the error:
```
curl: (56) OpenSSL SSL_read: error:1408F119:SSL routines:ssl3_get_record:decryption failed or bad record mac, errno 0
```
my raspbian version:
```
root@raspberrypi:~# lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
root@raspberrypi:~# 
```

pelwell, 2020-11-08:
Searching the web for the error message yielded this suggestion: 
```
$ sudo ethtool -K eth0 tso off gro off gso off ufo off
```

kamomil, 2020-11-08:
hi,
I tried `rpi-update` command severla times, at some point it succeded.
Can't say way. I actaully added `set -x` in the beggining of the `rpi-update` script to see what it runs,
the output of the succesfull execution was:
```
root@raspberrypi:~# rpi-update
+ REPO_URI=https://github.com/Hexxeh/rpi-firmware
+ REPO_API_URI=https://api.github.com/repos/Hexxeh/rpi-firmware
+ REPO_CONTENT_URI=https://raw.githubusercontent.com/Hexxeh/rpi-firmware
+ UPDATE_SELF=1
+ UPDATE_REPO_URI=https://github.com/Hexxeh/rpi-update
+ UPDATE_REPO_CONTENT_URI=https://raw.githubusercontent.com/Hexxeh/rpi-update
+ UPDATE_URI=https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
+ [[ unset == \u\n\s\e\t ]]
+ [[ unset != \u\n\s\e\t ]]
+ [[ unset != \u\n\s\e\t ]]
+ [[ unset == \u\n\s\e\t ]]
+ NOOBS_CHECK=1
+ BRANCH=master
+ ROOT_PATH=/
++ cut -d= -f3
+++ vcgencmd get_config os_prefix
++ echo =os_prefix=
+ CUR_FW_SUBDIR=/
+ FW_SUBDIR=/
+ [[ / != '' ]]
+ [[ / != \/ ]]
+ [[ / == \/ ]]
+ FW_SUBDIR=
+ BOOT_PATH=/boot
+ WORK_PATH=//root
+ SKIP_KERNEL=0
+ SKIP_SDK=0
+ SKIP_VCLIBS=0
+ SKIP_REPODELETE=0
+ SKIP_BACKUP=0
+ SKIP_DOWNLOAD=0
+ SKIP_WARNING=0
+ SKIP_CHECK_PARTITION=0
+ WANT_SYMVERS=0
+ WANT_PI4=0
+ PRUNE_MODULES=0
+ RPI_UPDATE_UNSUPPORTED=0
+ JUST_CHECK=0
+ RPI_REBOOT=0
+ CURL_OPTIONS=
+ GITHUB_API_TOKEN=
+ FW_REPO=https://github.com/Hexxeh/rpi-firmware.git
+ FW_REPOLOCAL=//root/.rpi-firmware
+ FW_PATH=/boot
+ FW_MODPATH=//lib/modules
+ FW_REV_IN=
+ FW_REVFILE=/boot/.firmware_revision
+ SELFUPDATE_SCRIPT=//root/.updateScript.sh
+ '[' 0 -ne 0 ']'
+ command -v vcgencmd
+ vcgencmd get_config str
+ grep -qE '^kernel='
+ CURL_OPTIONS=' -L'
+ CURL_OPTIONS_API='-A curl'
+ [[ -n '' ]]
++ eval curl -A curl -s -L https://api.github.com/rate_limit
++ tr -d ,
+++ curl -A curl -s -L https://api.github.com/rate_limit
++ awk 'BEGIN {reset=0;} { if ($1 == "\"limit\":") limit=$2; else if ($1 == "\"remaining\":") remaining=$2; else if ($1 == "\"reset'
+ GITHUB_API_LIMITED=0
+ '[' 0 -gt 0 ']'
+ [[ 0 -ne 0 ]]
+ echo ' *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom'
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
+ [[ ! -d //root ]]
+ [[ 1 -ne 0 ]]
+ update_self
+ echo ' *** Performing self-update'
 *** Performing self-update
+ local _tempFileName=/usr/bin/rpi-update.tmp
+ eval curl -fs -L --output /usr/bin/rpi-update.tmp https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
++ curl -fs -L --output /usr/bin/rpi-update.tmp https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update
++ head -1 /usr/bin/rpi-update.tmp
+ [[ #!/bin/bash != \#\!\/\b\i\n\/\b\a\s\h ]]
++ stat -c %a /usr/bin/rpi-update
+ local OCTAL_MODE=755
+ chmod 755 /usr/bin/rpi-update.tmp
100  117M  100  117M    0     0  2013k      0  0:00:59  0:00:59 --:--:-- 2783k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 5.4.74+
 *** depmod 5.4.74-v8+
 *** depmod 5.4.74-v7l+
 *** depmod 5.4.74-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to ece275eef974bb841f27a5f6353453b3b3c6644c
 *** A reboot is needed to activate the new firmware
```

Ruffio, 2020-11-21:
IMHO this can be closed

Ruffio, 2020-12-25:
@kamomil can this be closed?

kamomil, 2020-12-25:
yes

Ruffio, 2020-12-25:
@kamomil ok, then please close this issue. Thanks :-)

closed 2020-12-27 by kamomil