Hexxeh/rpi-update #311: rpi-update on rpi4/8G results in soft FP libs installed

pascal71, 2020-11-18:
Good afternoon,

While using rpi-update to update to the latest firmware or a beta firmware on a rpi4 on 64bit RPIos I see that rpi-update decides to install the software FP libs. I don't think that's a correct decission.

1) Can this be fixed?
2) What is the best way to install the HW FP libs? 

If needed I can help in further debugging/fixing the issue. Just let me know.

Kind regards,

    Pascal van Dam

popcornmix, 2020-11-18:
You are right the detection of hard/soft float fails on 64bit.
But both of these are 32-bit libraries, so fixing this would have no benefit.

There are no current plans to ship 64-bit libs with rpi-update
(the 64-bit OS stores these libs in a different location anyway).

In general these libs rarely change so updating isn't usually essential.
You can get the updated libs by rebuilding from userland repo.

pascal71, 2020-11-18:
Good afternoon,

Many thanks for the quick answer. So in short, running 64-bit Linux I am 
not running the risk of running FP tasks on the main ALUs instead of the 
FPUs?

Kind regards,

    Pascal van Dam

popcornmix, 2020-11-18:
Correct. 32-bit libs can't be used at all on a 64-bit OS so there is no danger they are having an effect.

pascal71, 2020-11-21:
Not a real issue on ARM64 deploys.

closed 2020-11-21 by pascal71