Hexxeh/rpi-update #314: Boot fails in encrypted sdcard after running `rpi-update`

jzazo, 2021-04-22:
Hi! I recently got an RP4 and was experimenting with it and trying LUKS encryption following the steps described in this pot:
https://www.raspberrypi.org/forums/viewtopic.php?t=219867 

After completing the steps, with encrypted sdcard, RP4 could boot normally. I then ran `sudo rpi-update` as part of other set of [instructions](https://www.tomshardware.com/uk/how-to/boot-raspberry-pi-4-usb) and after reboot I would get the following error:
```
Cannot initialize device-mapper. Is dm_mod kernel module loaded?
```
I tried fixing it by rebuilding initramfs following this [procedure](https://raspberrypi.stackexchange.com/questions/108375/raspbian-dm-mod-missing-after-mkinitramfs-on-luks-encrypted-partition), but it did not work. I will fall back to not using rpi-update after encryption, but wanted report this error as suggested in the original thread.

pelwell, 2021-04-22:
Your forum post includes the following:
> Everything worked perfectly, until I did "sudo apt full-upgrade" and "sudo rpi-update"

and yet you omitted the "apt full-upgrade" part from this Issue...

jzazo, 2021-04-22:
You are right, thanks for the correction. The set of instructions I linked specified that, and `sudo apt update`.

pelwell, 2021-04-22:
Because you performed both steps you now can't tell which was the cause of the breakage.

rpi-update is just an app the downloads over a network interface and writes files to an SD card or USB device. Provided it downloads and writes files to completion, it can't be blamed for any resulting errors. If the firmware or kernel is bad then the fault lies with the firmware or kernel. If the SD card/USB device gets corrupted then the fault lies with the device or the kernel.

jzazo, 2021-04-22:
I will try to reproduce at the end of the day today doing the first command only and rebooting; then the second, if you think that is meaningful.

My thought was that as a new kernel is installed, the modules change and initramfs is not updated accordingly. Also, I thought the old kernel might be removed and that's why it can no longer boot. I also thought that `rpi-update` was the one responsible for updating the kernel, but I will confirm if in this occasion it is the case.

I have a very shallow understanding of what could be happening, though.

jzazo, 2021-04-23:
I reinstalled the system and repeated the steps. The boot definitely fails after running `sudo rpi-update`. I understand that the failure is not `rpi-update`'s fault, rather the kernel's, so I am happy to close this issue and reopen where it corresponds. Is the raspberry's kernel [repo](https://github.com/raspberrypi/linux/issues) the right place to do this? Any idea of what is the issue also? Thanks.

closed 2021-04-25 by jzazo