Hexxeh/rpi-update #316: HDMI output hasn't worked since 78c1429cc1d5a200d824d2629c3ceba4ba4617fe

sheikki, 2021-05-02:
In the title the last release where I get HDMI output from my 8GB rPi 4. Using 4k resolution, don't know if that's relevant or not

pelwell, 2021-05-02:
Which 4K resolution, exactly?

sheikki, 2021-05-02:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=16

hdmi_enable_4kp60=1
gpu_mem=256
initial_turbo=30

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800
over_voltage=6
arm_freq=2000
v3d_freq=750

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=1

[all]
#dtoverlay=vc4-fkms-v3d
```

It's 3840x2160. Anything past the above release the display doesn't get a signal at all..

pelwell, 2021-05-02:
Can you confirm that removing the overclocks and gpu_mem makes no difference?

sheikki, 2021-05-03:
It seems like anything above 550 (which I believe is the default?) in `v3d_freq` leads to no HDMI signal to display..

popcornmix, 2021-05-03:
@sheikki all gpu (core, v3d, h264, hevc, isp) clocks share a pll now.
Your v3d overclock forces the PLL to 3GHz (with a divide by 4) which doesn't allow 550MHz for core with an integer divider.

Try using gpu_freq=750 (or whatever the highest value is that works). That should ensure core exceeds 550MHz.

sheikki, 2021-05-03:
But what about this part:

> It is recommended when overclocking to use the individual frequency settings (isp_freq, v3d_freq etc) rather than gpu_freq, as since it attempts to set core_freq (which cannot be changed on the Pi 4), it is not likely to have the desired effect.

From the [overclocking documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md)

pelwell, 2021-05-03:
> Or is the shared pll thing you mentioned relevant?

popcornmix is responsible for the patch that made the PLL more shared. You can assume the comment is relevant.

popcornmix, 2021-05-03:
I don't believe the change in PLL behaviour has reached stable/apt which is what the documentation describes.
It should be updated when this reaches apt. core_freq can now be changed in the same manner as other gpu clocks,
but only the highest of the frequencies will be accurate, the others are "best that can be done from that pll".
Because of that using gpu_freq is likely the safest option.

sheikki, 2021-05-03:
Ok so `gpu_freq=750` seems to work nicely with my unit. I'm not so sure it actually does anything though, but that might have been the case with the old `v3d_freq` line as well

```
while true; do vcgencmd measure_clock {core,v3d}; sleep 2; done
frequency(1)=533342272
frequency(46)=550006336
frequency(1)=550006336
frequency(46)=550006336
frequency(1)=533342272
frequency(46)=549993152
frequency(1)=533329088
frequency(46)=549993152
frequency(1)=533329088
frequency(46)=550006336
frequency(1)=550006336
frequency(46)=549993152
frequency(1)=533342272
frequency(46)=549993152
frequency(1)=533329088
frequency(46)=550006336
frequency(1)=549993152
frequency(46)=550006336
..
```

closed 2021-05-03 by sheikki