Hexxeh/rpi-update #318: mismatch of "uname -m" output and "getconf LONG_BIT"

uidn0158, 2021-06-21:
My original raspberry pi operating system is aarmv71, and I updated to aarch64 on local raspberry pi using following steps, after reboot, I use uname -m to check that it means my update is success, but it is still 32 using getconf LONG_BIT, Does anyone know what is reason? thanks.

-> sudo apt-get update
-> sudo apt-get upgrade
-> curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
-> sudo -i
-> mkdir /root/.rpi-firmware
-> cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
-> cp -r ./rpi-firmware-master/* ./
-> UPDATE_SELF=0 SKIP_DOWNLOAD=1 rpi-update
-> reboot
-> uname -a 
Linux raspberrypi 5.10.43-v8+ #1425 SMP PREEMPT Wed Jun 16 18:03:16 BST 2021 aarch64 GNU/Linux
-> getconf LONG_BIT
32

pelwell, 2021-06-21:
`getconf` shows the configuration of the Debian _OS_. What you have done is:
1. update to the latest version of your installed OS (presumably Raspberry Pi OS) which would have changed the kernel,
2. downloaded a snapshot of the firmware repo, including a new kernel,
3. moved the files around a bit,
4. installed it without downloading it using `rpi-update` (why not just use `sudo rpi-update` and skip steps 2 and 3?).

At some point you must also have set `arm_64bit=1` in config.txt. Without that you would still be running a 32-bit OS on a 32-bit kernel. As it is you are running a _32-bit OS on a 64-bit kernel_, which is why `getconf LONG_BIT` returns 32.

You can't upgrade to a 64-bit ARM OS from a 32-bit ARM OS, just as you can't upgrade from an X86_64 OS to an 32-bit ARM OS - in both cases you need to start from scratch.

uidn0158, 2021-06-21:
Hello pelwell, 

I added "arm_bit64=1" in end of /boot/config.txt, and I use uname -a to get aarch64 which means it is 64-bit kernel running, am I right? 
The reason I skip to use sudo rpi-update directly  as I got an error  "!!! Failed to download update for rpi-update!
!!! Make sure you have ca-certificates installed and that the time is set correctly", and I can't perform rpi-update even I re-install ca-certificates and sync. the time.

How I can fix my issue? re-install the system 2020-05-27-raspios-buster-arm64.zip using imager? or whether I could update to 64-bit ARM OS from 64-bit kernel?

Thanks for your supportl

Tim

pelwell, 2021-06-21:
Although the 32- and 64-bit OSes can both be run on the same device they are completely different architectures - you have to start from a fresh image download.