Hexxeh/rpi-update #319: cannot stat '//root/.rpi-firmware/*.elf' issue

johncblacker, 2021-06-26:


pelwell, 2021-06-26:
Don't make us guess - use `history` and post a list of the commands you ran.

johncblacker, 2021-06-26:
Sorry...
So, here are the commands I ran - all were run after: sudo su
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
SKIP_DOWNLOAD=1 rpi-update

The was done after about 5 attempts at just issuing:
sudo rpi-update
which failed each time...seemingly it couldn't download the firmware.  It would run a bit, then stop, then run a bit and stop, etc. and finally it just quit with the following messages:
curl: (18) transfer closed with 8726700 bytes remaining to read

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now

And, that's why I was trying to do the download - which worked perfectly without any of the stopping/starting I experienced when running the 
sudo rpi-update
command.

johncblacker, 2021-06-26:
Before I tried the rpi-update, I did this:
sudo apt-get update && sudo apt full-upgrade
and verified the pi would boot afterwords before I attempted the rpi-update 
process.  However, it's worth noting that after the attempts to run rpi-update, I could no longer boot this usb-stick...I'm starting to think that maybe I shouldn't be running the rpi-update at all, but my goal is to be able to netboot this pi and I found several of the netboot setup instructions recommended performing the rpi-update.

johncblacker, 2021-06-26:
At this point, I've reimaged my USB-stick and will do the 
sudo apt-get update && sudo apt full-upgrade
then once I verify the pi boots off the USB, I'll take the image over to my vbox/pi-server virtual machine, load it and give it a whirl!

popcornmix, 2021-06-28:
So where are you getting instructions from to do:
```
curl -L https://github.com/Hexxeh/rpi-firmware/archive/master.tar.gz -o master.tar.gz
cd /root/.rpi-firmware && tar -xvzf /root/master.tar.gz
```
That will extract files into `/root/.rpi-firmware/rpi-firmware-master` which isn't where rpi-update will look.
The script rpi-update effectively uses 
```
tar -xvzf --strip-components=1 /root/master.tar.gz
````
to remove the unwanted subdirectory, so try that.

johncblacker, 2021-06-29:
Thanks, I'll try that. obviously  theorists problem has to do with network or connection to github. I was able to get it to work yesterday  without errors. I will try your solution  next time I have the issue. Fwiw, I got those cmds off internet from back in 2018. Don't  have link, but it was in issues, maybe firmware repo?

johncblacker, 2021-06-29:
Found it. Closed issues #253 perrycannon, March 16, 2018

popcornmix, 2021-06-29:
I don't think that comment has ever been correct.
The post after I mentioned rpi-update runs with ``--strip-components=1``.
Without that you are extracting into an unexpected subdirectory.

johncblacker, 2021-06-29:
I am sure you are correct, especially since it didn’t work when I tried it.  I always take what I see online with a grain of salt and try to apply what knowledge I have 

against what’s posted, but in this case my pi experience is limited and rpi-update experience the same.  When I went down through the comments in that issue, I saw 

many that purported to work, but the one noted was the only one that seemed to me like it “should” work, but I’m lacking knowledge of rpi-update itself and how it 

works so I couldn’t try to fix the problem myself.  I might have been able to go through the rpi-update script and figure it out, but I chose what I thought was the easier

and more sensible approach to me, based on my knowledge.  Thank you for your suggestions and advice.  It’s interesting to watch the download when run via the script and

doing it directly via cmdline on the terminal; the later cranks along quite fast while the script version of the download stops/starts numerous times.  I got lucky the other

day and it ran to completion, so I’m going to file your suggestion in case I need it in the future.  Thanks again.

closed 2021-06-29 by popcornmix