Hexxeh/rpi-update #323: How would you upgrade/downgrade/restore the current stable "default" firmware?

trickkiste, 2021-12-27:
The default behaviour of rpi-update should be, to only update to the latest approved stable version, known to not introduce any bugs. It seems this can currently be achieved by setting BRANCH=stable, but this should be the default behaviour and needs to be stated somewhere explicitly.

The normal user, as I am, currently does not really understand, what this tool is for, how much one can f*** up their system with it and neither, how to get back to normal again. 

Does reflashing the SD card work?
Do I need to sudo BRANCH=stable rpi-update as well?
Is the "stable" state well tested?

I believe I have screwed up my bluetooth setup with rpi-update on my RPI 3. Or does it not work for everybody and only got fixed in RPI 3+?

trickkiste, 2021-12-27:
Is the firmware only put on the SD and loaded on boot each time (ergo reflashing the SD fixes everything) or is some firmware actually flashed onto the RPI chip itself?

That probably is the biggest question, which currently is completely unanswered in the documentation or at least not presented in the prominent place, such vital information should be placed at.

pelwell, 2021-12-27:
Read the [README](https://github.com/Hexxeh/rpi-update).

closed 2021-12-27 by pelwell