Hexxeh/rpi-update #327: Input output error

everdrone, 2023-02-17:
The updater fails with the following error:

```
% sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:f524729d84d527efd3751679c8c68445bb171b20
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
cp: error reading '/boot/kernel.img': Input/output error
```

if i manually write `sudo cp /boot/kernel.img test.cp` i get the same error

popcornmix, 2023-02-17:
Unrelated to rpi-update. Sounds like your sdcard is probably corrupt.
Does `dmesg` show any related errors?

everdrone, 2023-02-21:
Ah yeah a bunch actually
```
blk_update_request: I/O error, dev mmcblk0, sector 18596 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
```

I'll see if I can manage to replace it while keeping the data
Thank you

closed 2023-02-21 by everdrone